neutrons/addie #29: Reloading tab2 table fails

JeanBilheux, 2016-11-17:
Post-Processing Tab -> Reload List of Scans button.

fails for iPTS-16826 for example

marshallmcdonnell, 2016-11-18:
[image]

JeanBilheux, 2016-11-18:
Marshall, did you have the chance to try with an old autoNOM file and see if the same error pops up ?

marshallmcdonnell, 2016-11-18:
Sorry Jean, I just checked.

It does work for an older IPTS (IPTS-16915). The file that is being read-in is auto_sum.inp in the older IPTS. Is this file created by FastGR?

Below, I have the auto_sum.inp files for both IPTSs (16915 on the right and 16826 on the left):
[image]

JeanBilheux, 2016-11-18:
Thanks. That will help me debug the issue. I will try to work on it today (just pretending to list to the tutorial we have all day long ☺  ).

Those 2 IPTS will really help me figure out the issue. Thanks for doing this

Jean

[creenshot 20]<https://cloud.githubusercontent.com/assets/17032543/20433637/930f4498-ad72-11e6-9c9d-e6c1257e2794.png>

marshallmcdonnell, 2016-11-18:
It was not on your side! It was in the python script readtitles.py. 

It works now! Sorry about that and hope. I should have looked into it more first.

closed 2016-11-18 by marshallmcdonnell

JeanBilheux, 2016-11-18:
I never had such an easy bug to fix   ☺

thanks