neutrons/addie #70: autoNOM Tab - add copystuff.py capability into autoNOM Tab

marshallmcdonnell, 2016-12-20:


JeanBilheux, 2016-12-27:
arguments are:
-i <input IPTS>
-f <input autoNOM>
-t <output autoNOM>

JeanBilheux, 2016-12-27:
Can you explain to me how the arguments are supposed to work because when I give him the following folders (which I thought was the way it is supposed to work)...it does not like it and try to go to some very strange IPS-9999 folder !!!!

```
/SNS/users/zjn/pytest/copystuff.py -i /SNS/NOM/IPTS-17210 -f /SNS/NOM/IPTS-17210/shared/autoNOM_02 -t /SNS/users/j35/autonNOM_remove_me

mkdir: cannot create directory /SNS/NOM/IPTS-9999/shared//SNS/users/j35/autoNOM_remove_me/': No such file or directory 
.......etc.....
```

marshallmcdonnell, 2016-12-27:
The usage for the listed above should be as follows:

`/SNS/users/zjn/pytest/copystuff.py -i 17210 -f autoNOM_02 -t autoNOM_remove_me`

closed 2016-12-27 by JeanBilheux