neutrons/addie #141: Fix naming for tabs in modules

marshallmcdonnell, 2019-02-15:
Proposed name changes for better structure:
 * `addie.step1` -> `addie.autoNOM`
 * `addie.step2` -> `addie.processing.idl`
 * `addie.master_table` -> `addie.processing.mantid`
 * "loose scripts" -> `addie.rietveld`
 * "loose scripts" -> `addie.calculate_gr`

JeanBilheux, 2019-02-18:
In order to work on the loose files, I will need some instructions on how to load the data in those tabs and test the various features.

marshallmcdonnell, 2019-02-18:
If you go to a test ipts then you can look in the SofQ directory for the calculate gr tab and you can go into the GSAS directory for the Bragg Peak tab. You will use the Load S (Q) an Load Bragg buttons respectively and automatically visualize data.

I can also try the branch if you need me to help test

JeanBilheux, 2019-02-18:
Any good test IPTS? Is 21894 a good one?

marshallmcdonnell, 2019-02-18:
Sorry away from the computer right now. And without looking into it I couldn't really tell. But I do know ipts 19393 is good to go

JeanBilheux, 2019-02-18:
no problem. Enjoy your day/trip.

closed 2019-03-19 by peterfpeterson