neutrons/addie #150: Refactor of scripts that need to be split up

marshallmcdonnell, 2019-02-21:
Loose file that need to be split up:
 - [ ] `from addie.treelib import BraggTree` -> `addie.rietveld.braggtree.py`
 - [ ] `from addie.treelib import GofRTree` -> `addie.calculate_gr.gofrtree.py`
 - [ ] `from addie.graphicsviewlib import BraggView` -> `addie.rietveld.braggview.py`
 - [ ] `from addie.graphicsviewlib import GofRView` -> `addie.calculate_gr.gofrtree.py`

We need to look closer at the following since may go away with other issues:
 - [ ] `addie.addiedriver.py` - Definitely related to #144
 - [ ] `addie.ipythondockwidget.py` - See #143
 - [ ] `addie.mantid_ipython_widget.py` - See #143

And finally, a major refactor of `addie.addie` main script

JeanBilheux, 2019-03-07:
work will be done in a new branch called refactor_split_up

JeanBilheux, 2019-03-07:
Fix flake8 (82b0a7c903b387e50cb393d6213fed105a0f8e3c)

closed 2019-03-26 by peterfpeterson