neutrons/addie #192: Color/style issue in Rietveld tab

JeanBilheux, 2019-04-01:
Changing the style of the plot using the **color/style** button in Rietveld crashes the program.

marshallmcdonnell, 2019-04-05:
Errors with:
```
***** "s (square        )"
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/matplotlib/lines.py", line 1107, in set_linestyle
    ls = ls_mapper_r[ls]
KeyError: 's'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "build/scripts-3.6/addie", line 655, in do_set_bragg_color_marker
    rietveld_event_handler.do_set_bragg_color_marker(self)
  File "/home/ntm/direnv/python3/addie/build/lib/addie/rietveld/event_handler.py", line 469, in do_set_bragg_color_marker
    markercolor=color)
  File "/home/ntm/direnv/python3/addie/build/lib/addie/plot/mplgraphicsview.py", line 444, in updateLine
    self._myCanvas.updateLine(ikey, wkspname, wkspindex, linestyle, linecolor, marker, markercolor)
  File "/home/ntm/direnv/python3/addie/build/lib/addie/plot/canvas.py", line 419, in updateLine
    line.set_linestyle(linestyle)
  File "/usr/lib/python3/dist-packages/matplotlib/lines.py", line 1112, in set_linestyle
    "valid values.").format(ls))
ValueError: You passed in an invalid linestyle, `s`.  See docs of Line2D.set_linestyle for valid values.
```

JeanBilheux, 2019-04-10:
File "build/scripts-2.7/addie", line 649, in do_set_bragg_color_marker
    rietveld_event_handler.do_set_bragg_color_marker(self)
  File "/SNS/users/j35/git/addie_dev_in_progress/build/lib/addie/rietveld/event_handler.py", line 458, in do_set_bragg_color_marker
    plot_id_list, color, marker = ps.get_plots_color_marker(main_window, plot_label_list=plot_id_label_list)
  File "/SNS/users/j35/git/addie_dev_in_progress/build/lib/addie/utilities/specify_plots_style.py", line 169, in get_plots_color_marker
    child_window.set_plot_labels(plot_labels=plot_label_list)
  File "/SNS/users/j35/git/addie_dev_in_progress/build/lib/addie/utilities/specify_plots_style.py", line 132, in set_plot_labels
    assert len(plot_labels) > 0, 'Input plot lines cannot be an empty list.'
AssertionError: Input plot lines cannot be an empty list.

marshallmcdonnell, 2019-07-16:
Currently working on `master` branch.

Closing since appears to be fixes. Feel free to reopen if I am wrong.

closed 2019-07-16 by marshallmcdonnell