neutrons/addie #196: Rietveld tab. Right click in workspace table has some issues

JeanBilheux, 2019-04-01:
It's not possible to delete a workspace for example.

marshallmcdonnell, 2019-04-05:
Will raise the error:
```
Traceback (most recent call last):
  File "/home/ntm/direnv/python3/addie/build/lib/addie/rietveld/braggtree.py", line 256, in do_delete_gsas
    self._mainWindow.get_workflow().delete_workspace(ws_name)
  File "/home/ntm/direnv/python3/addie/build/lib/addie/addiedriver.py", line 168, in delete_workspace
    raise RuntimeError('Workspace %s does not exist.' % workspace_name)
RuntimeError: Workspace Bank 1 - 15.099999999999994 does not exist.
```

marshallmcdonnell, 2019-04-05:
Also, when trying to "Remove from plotting" on right-click, get the following:
```
Traceback (most recent call last):
  File "/home/ntm/direnv/python3/addie/build/lib/addie/rietveld/braggtree.py", line 213, in do_remove_from_plot
    gss_bank_names)
  File "/home/ntm/direnv/python3/addie/build/lib/addie/calculate_gr/event_handler.py", line 725, in remove_gss_from_plot
    bank_id = int(gss_bank_ws.split('_bank')[-1])
ValueError: invalid literal for int() with base 10: 'Bank 1 - 15.099999999999994'
```

marshallmcdonnell, 2019-04-10:
When we right-click to Plot the individual banks, we get:

JeanBilheux, 2019-04-10:
When we right-click to IPython, we get the following error
[image]

Kvieta1990, 2021-06-25:
issue fixed with PR [image]

closed 2021-06-25 by Kvieta1990