neutrons/addie #208: Calculate G(r) - Save G(r) is broken

marshallmcdonnell, 2019-04-03:
**To Reproduce**:
 - Grab a G(r) using Load G(R)
 - click Save G(r) button

Will error out with below:

JeanBilheux, 2019-04-03:
```
Traceback (most recent call last):
  File "build/scripts-2.7/addie", line 793, in do_save_gr
    calculategr_event_handler.do_save_gr(self)
  File "/SNS/users/j35/git/addie_dev_in_progress/build/lib/addie/calculate_gr/event_handler.py", line 385, in do_save_gr
    gr_file_name = get_file_names_from_dialog(default_dir, file_filter_str, caption)[0]
TypeError: 'NoneType' object has no attribute '__getitem__'
```

marshallmcdonnell, 2019-04-03:
If you do select a G(R) generated from S(Q), then it will let you save but if you select nothing and hit "Cancel" you get an error

marshallmcdonnell, 2019-04-03:
Also, if you select G(R) generated from S(Q) but don't put a valid file format, the GUI will crash and get the error below:

JeanBilheux, 2019-04-03:
```
  File "build/scripts-2.7/addie", line 793, in do_save_gr
    calculategr_event_handler.do_save_gr(self)
  File "/SNS/users/j35/git/addie_dev_in_progress/build/lib/addie/calculate_gr/event_handler.py", line 385, in do_save_gr
    gr_file_name = get_file_names_from_dialog(default_dir, file_filter_str, caption)[0]
TypeError: 'NoneType' object has no attribute '__getitem__'
Traceback (most recent call last):
  File "build/scripts-2.7/addie", line 793, in do_save_gr
    calculategr_event_handler.do_save_gr(self)
  File "/SNS/users/j35/git/addie_dev_in_progress/build/lib/addie/calculate_gr/event_handler.py", line 399, in do_save_gr
    raise RuntimeError('G(r) file with extension {0} is not supported.'.format(file_ext))
RuntimeError: G(r) file with extension  is not supported.
```

marshallmcdonnell, 2019-04-03:
Probably should add the extension instead of telling User it is the incorrect file format extension.

PR #259 (2019-07-16): Fix SavePDFGui for G(r) by setting label to angstrom
Fixes #208 

The issue was mostly fixed except the `SavePDFGui` function, which needed to have the x-axis unit label set to `Angstrom`. Added this fix.
closed 2019-07-16 by marshallmcdonnell