neutrons/addie #214: Post-Processing - Need to launch thread for TotalScatteringReduction

marshallmcdonnell, 2019-04-10:
We currently use the main thread for the GUI to launch reduction which causes the GUI to freeze during a running reduction (which can be ~15 minutes or more).

Also need to consider setting up a correct thread pool for flexibility in processing the jobs. Want the ability to process each row serially but should accommodate if we want to do more than 1 at a time.

marshallmcdonnell, 2019-04-10:
Need to refactor JobManager where we implement MVC (model view controller):
1) Top-level is a View that we show
2) Next level is a Controller 
3) Mantid controller needs to control our ThreadPool "Model", IDL will have a super simple Model where we put most of the current JobManager code.

Need to look into AsyncTask in Mantid for "underneath" the ThreadPool

peterfpeterson, 2019-04-10:
[This PR in mantid](https://github.com/mantidproject/mantid/pull/25492/files) added the results of executing the python script to `AsyncTask`.

marshallmcdonnell, 2019-04-10:
Here in `master`:
https://github.com/mantidproject/mantid/blob/master/qt/python/mantidqt/utils/asynchronous.py

And test for example:
https://github.com/mantidproject/mantid/blob/master/qt/python/mantidqt/utils/test/test_async.py

peterfpeterson, 2019-05-09:
The first round of the work is in #232

closed 2019-06-26 by marshallmcdonnell