neutrons/addie #216: Organization of addie.autoNOM

marshallmcdonnell, 2019-04-10:
We have code that runs for the Mantid reduction sitting in this module, specifically the `Launch Reduction` for Mantid reduction sits here.
https://github.com/neutrons/addie/blob/master/addie/autoNOM/event_handler.py#L40-L79

Need to move this to enforce the separation between the two and keep the code organized. 

@JeanBilheux do you have a suggestion where this should be moved and what else needs to change accordingly?

JeanBilheux, 2019-04-11:
because it's related to autoNOM, maybe we should keep it in this folder but inside a file called mantid.py.

marshallmcdonnell, 2019-04-12:
Actually, those lines of code I pointed out are for the new Mantid reduction and completely unrelated to autoNOM. I agree, needs to move to a separate file and then a different module.

JeanBilheux, 2019-04-18:
Any suggestions to where to put this code (location and name of module)

marshallmcdonnell, 2019-04-18:
Maybe in `addie/processing/mantid/master_table/` as `launch_reduction.py`? 

@peterfpeterson, any thoughts?

peterfpeterson, 2019-04-22:
What about as (I don't have a good filename) `addie/processing/mantid/launch_reduction.py`?

marshallmcdonnell, 2019-04-22:
@peterfpeterson That's sounds good to me :+1:

PR #243 (2019-06-28): Density processing of table JSON -> reduction-ready JSON
 - Density converters added
 - Moves common functions to `addie.utilities.math_tools`, specifically density converters and volume calculators
 - Fixes #216 so we have the `launch_recduction` for the mantid reduction in a sensible place
 - Debugging of density converter in UI so we update two of the unedited fields of mass density, number density, and mass (if chemical formula and geometry for mass field defined) if one of the others is edited.
 - Added a density processor to take our table JSON for `Density` section and replace it with a [... cut]
closed 2019-06-28 by marshallmcdonnell