neutrons/addie #222: Post-Processing - Introspection to find AlignAndFocusArgs

marshallmcdonnell, 2019-04-10:
Introspect into Mantid `AlignAndFocusPowder` algorithm to get parameters and create a drop-down in the Key-Value View under Key so we don't just allow free form entries that don't work.

peterfpeterson, 2019-04-10:
```python
alg = AlgorithmManager.createUnmanaged('AlignAndFocusPowderFromFiles')
alg.initialize()
propnames = [prop.name for prop in alg.getProperties()]
```
You probably want to blacklist the properties though as some are already spoken for in the table already (e.g. `Filename`), and some shouldn't be shown to the users (e.g. `UnwrapRef`, `LowResRef`, `LowResSpectrumOffset`).

marshallmcdonnell, 2019-04-10:
Oh good call, I forgot about the need to do that (ie black-list stuff)

marshallmcdonnell, 2019-04-12:
Blacklist from me
- Filename
- CacheDir
- OutputWorkspace
- CalFileName
- GroupFilename
- ResampleX
- Dspacing
- UnwrapRef
- LowResRef
- LowResSpectrumOffset

And just to check, the whitelist will then be:
- MaxChunkSize
- FilterBadPulses
- AbsorptionWorkspace*
- UnfocussedWorkspace
- Characterizations
- GroupingWorkspace*
- CalibrationWorkspace*
- OffsetsWorkspace*
- PreserveEvents
- RemovePromptPulseWidth
- CompressTolerance
- CompressWallClockTolerance
- CompressStartTime
- ReductionProperties
- DMin
- DMax
- TMin
- TMax
- CropWavelengthMin
- CropWavelengthMax
- PrimaryFlightPath
- SpectrumIDs
- L2
- Polar
- Azimuthal
- FrequencyLogNames
- WavelengthLogNames

*Ones I am "torn" on because they are input workspaces. Could create them in the IPython widget so I lean towards whitelisting these

@psqrd do you have any others that should be switched?

commit c456ed20 (2019-04-12): Updated blacklist. this closes #222
closed 2019-04-12 by JeanBilheux