neutrons/addie #262: Calculate G(r) - crash on right-click to plot a PDF

marshallmcdonnell, 2019-07-25:
Missing `plot_gr` method [here](https://github.com/neutrons/addie/blob/master/addie/calculate_gr/gofrtree.py#L706) 

Candidates are: 
1) method of GofRView class in `addie.calculate_gr.gofrview` 
https://github.com/neutrons/addie/blob/master/addie/calculate_gr/gofrview.py#L93

2) function in `addie.calculate_gr.event_handler`
https://github.com/neutrons/addie/blob/master/addie/calculate_gr/event_handler.py#L291

Probably got changed with refactor. Going to look at function call signature to figure out the proper one to use.

**Error**
```
Traceback (most recent call last):
  File "/home/ntm/direnv/python3/addie/addie/calculate_gr/gofrtree.py", line 706, in do_plot
    self._mainWindow.plot_gr(gr_list, None, None, None, auto=True)
AttributeError: 'MainWindow' object has no attribute 'plot_gr'
```

marshallmcdonnell, 2019-07-25:
Same for `plot_sq`

peterfpeterson, 2019-07-26:
`gofrview` has its own `plot_gr` method that can be called with the workspace name. Same goes for `sofqview`.

marshallmcdonnell, 2019-07-26:
@peterfpeterson Thanks, I got it sorted yesterday in the end and it is working in the[`262_plot_gr`](https://github.com/neutrons/addie/tree/262_plot_gr) branch.

No PR yet because I am fixing _insert curse words_ conda stuff in CI. Once I get that back up, will submit (with you as a reviewer? :grin: )

PR #264 (2019-07-30): FIxes `plot_gr` and `plot_sq` method calls
Fixes #262
closed 2019-07-30 by marshallmcdonnell